gijzelaerr/python-snap7 #196: Feature request: ReadSZL()

NicklasMD, 2020-09-25:
Hi I didn't know where the right place where to ask for this?
The Snap7 has a function to read SZL values or lists of values, I would like that also in the Python-Snap7
Thanks for all your work.

**Example from the Snap7 documentation:**
Cli_ReadSZL
**Description**
Reads a partial list of given ID and INDEX.
**Declaration**
int Cli_ReadSZL( S7Object Client, int ID, int Index,
TS7SZL *pUsrData, int *Size);
function Cli_ReadSZL(Client : S7Object; ID, Index : integer;
pUsrData : PS7SZL; var Size : integer) : integer;
**Parameters**
Type Dir.
Client Native Integer In
The handle as return value of
Cli_Create(), passed by value.
ID integer 32 In List ID
Index integer 32 In List Index
pUsrData Pointer to struct in Address of the user buffer
size Pointer to integer 32 In Buffer size available
Out Bytes rea

swamper123, 2020-09-25:
What is an SZL btw? Never heared of that thing in an SPS context before (or can't remember).

This feature is actually not implemented, but feel free to make a PR.
I would recommened that you check other functions/methods and adapt their "wrapping behaviour".

NicklasMD, 2020-09-25:
The SZL's are used to get diagnostic data out of the PLC
For example status on the fieldbus, diagnostic buffer, status LED's etc. so I think it's a quite useful feature.

I'm not sure I'm competent enough to make the feature myself...

swamper123, 2020-09-25:
Found out SZL means "SystemZustandsListe", so like you said a diagnostic datas.

I thought I may never contribute to this repo as well, but after a while I am pretty into it in any kind of way. 😄  
If you inspect other client requests of their structure, you'll see that they are pretty similar and you see where `self.library....` is used, that this is the actual C command.

Otherwise you may have to wait a longer time until somebody implements it or you have to make PLC workarounds.

lautarodapin, 2020-09-26:
I really would love that two, I was looking for something like that for checking if devices where connected 
Didn't know that szl was the option
I will check if i can't get to something

closed 2020-12-02 by gijzelaerr