gijzelaerr/python-snap7 #197: Obtain data block layout specification

lifeisawavesorideit, 2020-10-20:
Hello,

I've been reading the examples provided, especially those where one can upload a specific data block and provide its layout. In the utily.py following comments are made to the parameter _layout_:
# layout specification DB variable data
# A DB specification is the specification of a
# DB object in the PLC you can find it using
# the dataview option on a DB object in PCS7

I  have to admit, that my knowledge of S7 and TIA is very limited but can one directly export this layout specification to a text file? Or how are you able to provide the DB layout specification, where do you get it from?

spreeker, 2020-10-20:
Hi, 

I think i just copy pasted it into a text file from the exel/tabular like dialog you can see if you go into the details of a DB block.

lifeisawavesorideit, 2020-10-20:
Hi @spreeker,

oh, ok, that may be easier than predicted. I will definitely check it out.
Is there another way to obtain this layout specification? By that I mean with the python-snap7.

closed 2020-11-02 by gijzelaerr