gijzelaerr/python-snap7 #198: High frequency data acquisition

lifeisawavesorideit, 2020-10-21:
Hello,

I would like to gather data from the PLC (2-3 DBs each around 150-200 parameters) with a high frequency and store this data to a database. High frequency would be around something higher than 10 Hz, ideally 100Hz. 
What would be the best option to do that and is it possible? Does anyone have experience with the python-snap7 and high frequency data acquisition?

I am currently testing my code for one (1) DB with 20 parameters and I am achieving a frequency of 5 Hz but I could still improve my code so I think 10 Hz can be realistic. But what about 100 or even 1000 Hz? Is there a "streaming"-function / option?

spreeker, 2020-10-21:
Hi,  setting a single value takes about 5ms.  That's is a whole round trip. The 'clock' speed of a PLC is actually very low compared to modern cpu's. For a small amount to data 20 times a second cloud be possible. Using the s7 protocol. 

The speeds you are requiring maybe need a complete different approach with just your PLC send udp packets. to a programmed ip address. Which you have to make manually the sending code and receiving code.
But the (I assume) the sensors you are reading also have a update frequency.   You are bumping in the boundaries of what a PLC can do I think :).

gijzelaerr, 2020-11-02:
i guess this question has been answered.

closed 2020-11-02 by gijzelaerr