gijzelaerr/python-snap7 #199: reading from an S7-1200 - DB100.10 (real value)

stevennoppe, 2020-10-22:
In my PLC I have a DB100 from which I want to read from.

I have 4 variables in the DB100 : 
DB100.0 = real
DB100.4 = real
DB100.8 = int
DB100.10 = real

I can read and show the first 3 variables with my code, but when I try to read the last variable, I get an "address out of range" error. But I cannot seem to find out why I cannot read that last variable.

This is my code : 

`# function for reading the data easier
def ReadMemory(plc, db, byte, bit, datatype):
  result = plc.read_area(areas['DB'],db,byte,datatype)
  if datatype==S7WLBit:
    return get_bool(result,0,1)
  elif datatype==S7WLByte or datatype==S7WLWord:
    return get_int(result,0)
  elif datatype==S7WLReal:
    return get_real(result,0)
  elif datatype==S7WLDWord:
    return get_dword(result,0)
  else:
    return None

    #read DB100.0
    Machine_speed = ReadMemory(plc_slakbewaking,100,0,0,S7WLReal)

    #read DB100.4
    Roll_length = ReadMemory(plc_slakbewaking,100,4,0,S7WLReal)

    #read DB100.8
    Knife_3_level = ReadMemory(plc_slakbewaking,100,8,0,S7WLWord)

    #read DB100.10
    Knife_3_laser_value = ReadMemory(plc_slakbewaking,100,10,0,S7WLReal)
`
What's happening? :(

lautarodapin, 2020-10-22:
```py
# function for reading the data easier
def ReadMemory(plc, db, byte, bit, datatype):
  result = plc.read_area(areas['DB'],db,byte, byte + datatype) # <---- here
  if datatype==S7WLBit:
    return get_bool(result,0,1)
  elif datatype==S7WLByte or datatype==S7WLWord:
    return get_int(result,0)
  elif datatype==S7WLReal:
    return get_real(result,0)
  elif datatype==S7WLDWord:
    return get_dword(result,0)
  else:
    return None

    #read DB100.0
    Machine_speed = ReadMemory(plc_slakbewaking,100,0,0,S7WLReal)

    #read DB100.4
    Roll_length = ReadMemory(plc_slakbewaking,100,4,0,S7WLReal)

    #read DB100.8
    Knife_3_level = ReadMemory(plc_slakbewaking,100,8,0,S7WLWord)

    #read DB100.10
    Knife_3_laser_value = ReadMemory(plc_slakbewaking,100,10,0,S7WLReal)
```

stevennoppe, 2020-10-22:
Hello, Thx for the reply. I will try this Monday.
But I wonder why does it work for the first 2 real values and not for that last one?

lautarodapin, 2020-10-22:
I notice an error of my part.
If you are using the module types from snap7 (Ex: `S7WLReal`) those values are hexa, that is why it worked for some of the values but not for all of them.
you could use the ADict wordlen_to_ctypes to check for the length 
```py
read_area(AreaType, number, 0, end + offset) 
```

gijzelaerr, 2020-11-02:
b

closed 2020-11-02 by gijzelaerr