gijzelaerr/python-snap7 #203: test\test_client_async.py fails on windows

gijzelaerr, 2020-10-28:
```Run pytest.exe
============================= test session starts =============================
platform win32 -- Python 3.7.9, pytest-6.1.1, py-1.9.0, pluggy-0.13.1
rootdir: D:\a\python-snap7\python-snap7
collected 172 items

test\test_client.py .s.sss.ssssssssss........................s.......... [ 30%]
...........................................                              [ 55%]
test\test_client_async.py Entering debug mode. Use h or ? for help. 

At D:\a\_temp\8ec68c21-4c58-41b6-9035-e080ce31506e.ps1:3 char:5
+ if ((Test-Path -LiteralPath variable:\LASTEXITCODE)) { exit $LASTEXIT …
+     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[DBG]: PS D:\a\python-snap7\python-snap7>> 
Error: Process completed with exit code 1.
```

https://github.com/gijzelaerr/python-snap7/runs/1319367971?check_suite_focus=true

swamper123, 2020-10-28:
The client_async.py and test_client_async stuff will be revised in near future anyway, when the as_* stuff in client.py is working and accepted. There are known problems with asyncio <=3.7 and windows machines. I will have that in my mind while working on the asyncio stuff later.

swamper123, 2020-10-28:
If this is hinder your test here, you can delete that test folder and the client_async.py as well.
They both has to be revised anyway, when the as_* stuff in client.py is working and accepted. I got a branch for that purpose based on [61c970c0](https://github.com/gijzelaerr/python-snap7/commit/61c970c09e5ce1a2c30b1ed3e71fe16f19ffb527).

swamper123, 2020-10-28:
The same thing happens in test_partner as well:

https://github.com/gijzelaerr/python-snap7/pull/200/checks?check_run_id=1321783239

swamper123, 2020-10-29:
Why does test_partner doesn't need a tearDown to stop Partner()? 
Could this cause some trouble here?

swamper123, 2020-11-02:
So it realy seems to be a "Server in Background" thing as mentioned in #206 

In the experimental draft #205 the windwos tests are now running ... but now the python tests fail. 3.6 and 3.7 is fine.
Python 3.8 is just crashing without any hint and Python3.9 has an assertion error, I can't understand.

@gijzelaerr Do you know if there is a possibility to increase the space for logs (some Pytest summaries are cut) and if there is a possibility to gain more information from github actions in case something crashes? Because `Error: The operation was canceled.` is not very helpfull. 😕

gijzelaerr, 2020-11-02:
to me it just sounds still that the C lib is doing an illegal mem operation, the process SEGFAULTs en gives that error.

it can be quite painful to work with C libs from a python perspective since it can give all kind of weird behaviour when things go wrong.  That's why you want to have that all centralised, error checked and caged in a box with lasers and security beams so nothing scary comes in or out.

swamper123, 2020-11-09:
If somebody wants to join the discussion in the snap7 forum about possible Thread issues [click here](https://sourceforge.net/p/snap7/discussion/general/thread/6c45f37463/?limit=25#8723). Perhaps related with #26 as well.

gijzelaerr, 2020-11-30:
this is hopefully resolved in https://github.com/gijzelaerr/python-snap7/pull/213

closed 2020-11-30 by gijzelaerr