gijzelaerr/python-snap7 #207: Not able to write a single bit with consistency

JCirera, 2020-11-03:
I am not able to write a single bit to the PLC without re-writing the entire byte. I did not find the function implemented and I have tried the following code without success. I will greatly appreciate some insights.

Let's imagine that I would like to write the bit DB60.DBX5.3. The implemented code is the following:

```python
import snap7

cli=snap7.client.Client()
cli.connect('xxx.xxx.xxx.xxx',0,0)
type_ = snap7.snap7types.wordlen_to_ctypes[snap7.snap7types.S7WLByte]  
data = bytearray([1])                          
amount = 1
cdata = (type_ * amount).from_buffer_copy(data) 
snap7.common.load_library().Cli_WriteArea(cli.create(),snap7.snap7types.S7AreaDB, 60, 43, 1, snap7.snap7types.S7WLBit, ctypes.byref(cdata))
```
This code does not throw any error message. However, it does not work. I did not find the solution anywhere. 

Thanks in advance.

spreeker, 2020-11-03:
Well the minimum size you send to a plc is a byte so to change a bit in a
byte you need to send the entire byte..

Stephan

JCirera, 2020-11-04:
Issue closed as it seems that it is not possible.

closed 2020-11-04 by JCirera

swamper123, 2020-11-04:
@JCirera The only way would be, reading the Byte which contains the bit, change the bit, write the byte back. The easiest way I come up so sudden.

JCirera, 2020-11-04:
Thanks for the suggestion @swamper123. However, this solution is not consistent as:

- Step 1: read byte
- Step 2: modify bit inside byte
- Step 3: write byte.

Between steps 1 and 3, the PLC could have changed the value of the byte. Therefore, this solution is overwriting the entire byte without taking into account this scenario.

gijzelaerr, 2020-11-04:
Did you see this thread?:

https://sourceforge.net/p/snap7/discussion/general/thread/b4c7b4f4/

"The second method is consistent. You can use WriteArea passing S7WlBit as wordlength. The CP (communicator processor) changes only the desired bit."

@JCirera are you basing your code on this example?

JCirera, 2020-11-04:
Hi @gijzelaerr. Yes, I started from this thread, but it does not seem to work.

spreeker, 2020-11-04:
Hi @JCirera,

Well you can modify your plc program
By putting the entire bit in its own byte. Although not so memory efficient.

If you are worried about inconsistencies you need a different approach and
use current values set only by plc and read out only and have wished state
values where the plc applies or try to change the state and if successful
The current value == wished state. When dealing with a machine there is
some serious lag between setting a value and the actual change.

Using this makes sure you know what the plc is instructed to do.

Cheers,

Stephan.

jappoz, 2022-02-05:
Hello guys, 
for those who are still interested in this problem I eventually found out the correct way to write a single bit with consistency following the example provided in https://sourceforge.net/p/snap7/discussion/general/thread/b4c7b4f4/

My solution:

```
def write_bit_consistent(client, db_number, start, bit_index, value):
    _start_address = start * 8 + bit_index
    _type = snap7.types.wordlen_to_ctypes[snap7.types.WordLen.Byte.value]

    data = bytearray([0])
    if value:
        data = bytearray([1])

    cdata = (_type * 1).from_buffer_copy(data)

    res = client._library.Cli_WriteArea(client._pointer, snap7.types.S7AreaDB, db_number, _start_address, 1, snap7.types.S7WLBit, byref(cdata))
    snap7.common.check_error(res, context='client')
    return res
```

Example: Write DB1.DBX0.3 to true:

```
client = snap7.client.Client()
client.connect('192.168.0.1', 0, 0)

write_bit_consistent(client, 1, 0, 3, True)
```