gijzelaerr/python-snap7 #209: Snap7 for Raspberry pi 4 with ARM64

KHDOUDI, 2020-11-09:
Hello,
I would like to get your help about an issue during the setup of python snap7 on a Raspberry pi4 running Raspian OS Buster **ARM64** system.

i am using ARM64 because it's the only version allowing the setup of the Zebra scanner linux SDK (which is mandatory for my project).

But, for my Snap7 setup, i followed the steps for compiling snap on the Rpi4 trying the following:
......
......
`sudo make -f arm_**v7**_linux.mk install`

output:

```
g++ -c  "../../src/sys/snap_msgsock.cpp" -O3 -g -fPIC -mword-relocations -pedantic -o ../temp/arm_v7/sys_snap_msgsock.o -I. -I../../src/sys -I../../src/core -I../../src/lib 
g++: error: unrecognized command line option ‘-mword-relocations’; did you mean ‘-fno-resolution=’?
make: *** [common.mk:67: ../temp/arm_v7/sys_snap_msgsock.o] Error 1
```

but if i run:
`sudo make -f arm_**v6**_linux.mk install`
or
`sudo make -f **x86_64**_linux.mk install`
or even
`sudo make -f **i386**_linux.mk install`

the installation go without error... but the maybe it's not the suitable version because i continue having (as python error):

`Snap7Exception: can't find snap7 library. If installed, try running ldconfig`

which maybe mean by setup was not done correctly. is it related to my ARM64 version of raspbian ? if yes, what can be the solution to get Snap7 to work on this system.

Thank you for your support

swamper123, 2020-11-09:
Hey ho,

for the snap7 stuff, you may ask the developer [here](https://sourceforge.net/p/snap7/discussion/). I recommened to post the problem in _General Discussion_ , he answers mostly on the same date. Cool Dude.

Did you tried executing `ldconfig` and tried it again? Because this has to be done before using snap7 or python-snap7.

KHDOUDI, 2020-11-09:
Hello,

Thank you for your feedback, i just posted the same problem description on the Sourceforce.
Yes libsnap7.so is copied in correct path.
i tested running `ldconfig` and retry but the situation do not seems to improve.
Thank you

closed 2020-11-09 by KHDOUDI