gijzelaerr/python-snap7 #211: Issue with set_string

sm4xas, 2020-11-13:
Hello
I'm trying to set a string value i a DB.

This is the code I'm running (cannot get the indent to work properly on GitHub, sorry):

def writeString(plc,db,byte,bit,value):
data = bytearray(255)
    data[0]=0xfe
    set_string(data,0,value,255)
    plc.write_area(areas["DB"],db,byte,data)

writeString(plc,1,6,0,'Testar igen')

I'm getting this output:

Traceback (most recent call last):
  File "test_s7.py", line 61, in <module>
    writeString(plc,1,6,0,'Testar igen')
  File "test_s7.py", line 41, in writeString
    set_string(data,0,value,255)
  File "/home/dafo/.local/lib/python3.6/site-packages/snap7/util.py", line 204, in set_string
    _bytearray[byte_index + 2 + r] = ord(' ')
IndexError: bytearray index out of range

If I modify util.py and change the following, it works:
for r in range(i + 1, _bytearray[byte_index]):
        _bytearray[byte_index + 2 + r] = ord(' ')
to
for r in range(i + 1, _bytearray[byte_index]-1):
        _bytearray[byte_index + 2 + r] = ord(' ')

I get the same error if I read out the old string and modify it:

def writeString(plc,db,byte,bit,value):
    data = plc.read_area(areas['DB'],db,byte,255)
    set_string(data,0,value,255)
    plc.write_area(areas["DB"],db,byte,data)

Is this a bug or am I doing something wrong?

swamper123, 2020-11-17:
Hey @sm4xas 

> This is the code I'm running (cannot get the indent to work properly on GitHub, sorry):

Select your code and press this button:
[image]

It should make a codeblock then with the right indention etc.

Anyway: It seems that this is a function which hasn't got a test yet, so it might be broken. :|

As a workaround:
Could you please check if it works with `plc_value = set_value(0, "STRING", "Testar igen")` and try to write that value to the plc.
If that works, we may get a good starting point to get `set_string()` work again.

DominikK0702, 2020-11-19:
Hey @sm4xas

got the same issues with strings around a year ago. I can remember that the bytearray needs to be 2 bytes bigger than the data you wanna store, because the plc needs additional information about this datatype (current length of the string, max lenght of the string).

I hope i could help you with this issue.

Here a snippet of my working code.
```python    
def readString(self, db, start, len=255):
        if self.connect():
            reading = self.client.read_area(snap7.snap7types.S7AreaDB, db, start, len)
            return snap7.util.get_string(reading, 0, len)
        else:
            return self.readString(db, start, len)

def writeString(self, db, start, value):
        if self.connect():
            reading = self.client.read_area(snap7.snap7types.S7AreaDB, db, start, 255)
            snap7.util.set_string(reading, 0, value, 255)
            self.client.db_write(db, start, reading)
        else:
            return self.writeString(db, start, value)
```

pinoy4, 2020-11-26:
Hello. I have also encountered problems when using the `set_string` function. I found out that the issue at hand is that this function does not set the first byte with the max length of the string (as a reminder the default max length should be 254 if not specified). By manually setting the first byte to 254 in my case I was able to use this function without problems.

In @DominikK0702 's example he first reads the string from the PLC (so the first byte is included) and then writes his value on top of that without overwriting the first byte so in the end it works.

Now, my question is: is this the intended behaviour? It is very strange to me but I might be missing something.
I would gladly make a pull request adding this functionality if it is correct.

swamper123, 2020-11-27:
@pinoy4 I guess a method not working as expected, is not intended by anyone. ;)
The assert may be an issue in general, since it should be used in tests only. In production this may be bypassed and won't have an effect.

I used[ struct.pack() and "x{}p" as format](https://docs.python.org/3/library/struct.html) to convert string to binary...but many roads lead to Rome. 

If you would make an PR and a test for that function, it would be great.:)

MountainKing91, 2020-12-05:
Hi, 
I experienced weird behaviour too with the `set_string` and `get_string` method. I used this workaround: 

```
myplc = snap7.client.Client()
myplc.connect('127.0.0.1', 0, 1)

myint = 100 # byte 0, 1
myfloat = 123.45 # byte 2, 3, 4, 5
mystring = 'abstract' # byte 6, 7, 8, 9, 10, 11, 12, 13

data = struct.pack('>H', myint) + struct.pack('!f', myfloat) + mystring.encode()

myplc.db_write(1, 0, data) 
response = myplc.db_read(1, 0, 14)
print(response.hex(' '))

print(get_int(response, 0)) # myint
print(get_real(response, 2)) # myfloat
#print(get_string(response, 6, 8)) # mystring, but it doesn't get the first two letters
print(response[6:14].decode()) # mystring, this one works fine
```

gijzelaerr, 2021-02-05:
ive enabled 'discussions' for python-snap7, we kindly request to move any requests for help to this location https://github.com/gijzelaerr/python-snap7/discussions

closed 2021-02-05 by gijzelaerr