gijzelaerr/python-snap7 #215: Replace asserts which are not used in tests

swamper123, 2020-11-30:
In production, assert statements are ignored, if the python interpreter is set to optimised execution.
It is common to use them just in test cases.

gijzelaerr, 2020-11-30:
i never disable assert statements in production? Are you talking about `-O` flags? If you use those I guess you know what you doing and you are on your own.

swamper123, 2020-11-30:
In my case, I compiled Python with `--enable-optimisation` to gain an execution boost of ~15% while executing, which is a nice to have, especially if you have smaller devices.  This would e.g. remove asserts but does even more.

The -O flag is set before running the interpreter, which just removes the asserts afaik.

[The Python docs say](https://docs.python.org/3.8/reference/simple_stmts.html#assert):

> Assert statements are a convenient way to insert _debugging_ assertions into a program:
> 
>     assert_stmt ::=  "assert" expression ["," expression]
> 
> The simple form, assert expression, is equivalent to
> 
> if __debug__:
>     if not expression: raise AssertionError
> 

But if you say, that such cases belongs to the user, I will accept that and this issue can be closed.

gijzelaerr, 2020-11-30:
If speed is very important than python is the wrong language :)

afaik the way we use the asserts is just to make absolutely sure no logical errors are made. Ideally, an end-user never sees an assert.  Now i think about it for a while I do think it might be beneficial to improve the error reporting, but I don't think this really has a very high priority at the moment.

swamper123, 2020-12-01:
> If speed is very important than python is the wrong language :)

Yeah, I know...but I shall use it that way (command from above)... 😕
But you are right about the priority, but shall be listed here for the future. :)

gijzelaerr, 2021-03-03:
addressed in https://github.com/gijzelaerr/python-snap7/pull/250

closed 2021-03-03 by gijzelaerr