gijzelaerr/python-snap7 #235: Unable to read any DB or variable from S7-1200

niravpatidar37, 2021-01-08:
plc = snap7.client.Client()
plc.connect('192.168.1.101', 0,1) 
all_data = plc.db_get(1)
  File "C:\Python\lib\site-packages\snap7\client.py", line 234, in db_get
    check_error(result, context="client")
  File "C:\Python\lib\site-packages\snap7\common.py", line 66, in check_error
    raise Snap7Exception(error)
snap7.exceptions.Snap7Exception: b'CPU : Function not available'

PLC used :- S7-1200
CPU :- 1215C 
Firmware :-  4.4

Unable to get Datablock1 from the PLC

lautarodapin, 2021-01-18:
The function `db_get` will not  work on the plcs 1200 and 1500, you should use `db_read` or `read_area`
Also you should check in the plc program that put/get is allowed also that the db isn't optimized

gijzelaerr, 2021-02-05:
ive enabled 'discussions' for python-snap7, we kindly request to move any requests for help to this location https://github.com/gijzelaerr/python-snap7/discussions

closed 2021-02-05 by gijzelaerr