gijzelaerr/python-snap7 #243: Pythonize all internal objects

gijzelaerr, 2021-01-19:
The current struct wrappers can have a cleanup and made more pythonic.

lautarodapin, 2021-01-22:
Can you explain what you mean by cleanup and more pythonic? I want to contribute.

gijzelaerr, 2021-01-25:
hi @lautarodapin i don't have much time to go into depth here, but If you want to contribute you can also start with improving the docstrings of all methods:

https://github.com/gijzelaerr/python-snap7/issues/242

Just replicating the snap7 manual, adjusting the formatting would be of great help.

gijzelaerr, 2021-04-22:
I think we should replace the ADict objects with dataclasses. Note that this has been added in python 3.7, and we still support 3.6. So we need to add dataclass as a dependency.

gijzelaerr, 2024-05-02:
ADict is gone, no need for a dataclass in the end.

closed 2024-05-02 by gijzelaerr