gijzelaerr/python-snap7 #244: cannot import the snap7.dll

atalayismail16, 2021-01-19:
Hi i'am facing with the connecting to the plc with snap7 library does anyone have the true guide ?

swamper123, 2021-01-19:
http://snap7.sourceforge.net/

especially have a look on the [hardware compatibility](http://snap7.sourceforge.net/snap7_client.html#target_compatibility)

atalayismail16, 2021-01-19:
Everything is looking normal in hardware. I have s71200 plc i just installed snap7 library with pip and i'm using a anaconda virtual enviroment there are some topics about the solution with snap7.dll but none of them worked for me do you have any suggestion

lautarodapin, 2021-01-19:
Did you copy the dll to the system32 directory? 
I had a lot of problems with snaps in windows, try reinstalling anaconda or copy the dll to the python path as well

swamper123, 2021-01-20:
Like @lautarodapin mentioned, copy/pasting them in the system32 directory (or and other Enviroment Varaible of your Windows system).

Remember! PYTHONPATH != PATH

Some solutions are mentioned here:
Perhaps #56 #63 #187

Safest way is to copy paste the right them in the _Windows\system32_ folder (mostly listed in Path, unless somebody touched it and didn't know what he was doing). 

Check that you chose the right files from the snap7-full-1.4.2 folder->releases->Windows-> either Win32 or Win64 or when you got a realy old PC Legacy. _Not just .dll alone, copy all of them._

atalayismail16, 2021-01-28:
do you have any guide for install ?

swamper123, 2021-01-28:
tldr: https://python-snap7.readthedocs.io/en/latest/installation.html

tldr the tldr:

**Windows**:

1) Install snap7:

[Download snap7 from sourceforge](http://sourceforge.net/projects/snap7/files/)
from the snap7-full-1.4.2 folder->releases->Windows-> either Win32 or Win64 or when you got a realy old PC from Legacy. 
Not just .dll alone, copy all the files in the folder into  _Windows\system32_ (which should be there to 99%) or into another Folder which is in your Windows Enviroment Variables list in Path (in the Systemvariable list!).

2) $ pip install python-snap7

3) Should work!

**Linux:**

1) Install snap7:

fastest way:
$ sudo add-apt-repository ppa:gijzelaar/snap7
$ sudo apt-get update
$ sudo apt-get install libsnap7-1 libsnap7-dev

2) Install python-snap7:
fastest way:
$ pip install python-snap7

3) Make it work:
$ ldconfig

4) Should work now!

---
If not:
- check your python (virtual) enviroment of your project, did you install it there or somewhere else
- Windows: Check if all the files from step 1) are realy in one of the Path folders
- Linux: have you realy done step 3) ?
- Otherwise, you may ask again.

swamper123, 2021-02-04:
@atalayismail16 Did it work?

gijzelaerr, 2021-02-05:
ive enabled 'discussions' for python-snap7, we kindly request to move any requests for help to this location https://github.com/gijzelaerr/python-snap7/discussions

closed 2021-02-05 by gijzelaerr