gijzelaerr/python-snap7 #271: example/boolean.py and example/read_multi.py broken

gijzelaerr, 2021-04-22:
they don't match the snap7 anymore. Maybe we should add checking these files to github actions also.

swamper123, 2021-04-22:
We may should add/fix the examples in general. The code hasn't to be long, most stuff would be copy/paste, perhaps even stuff from the tests.
I would prefere multiple examples with less lines than one or few giant examples.

ToDos I think about could be:
- Client. read (add typical Datatype here) 
- Client read as (add typical Datatype here)
- Client. write (add typical Datatype here) 
- Client write async (add typical Datatype here)
- Client do sth. more complex...like db_fill or db_get or handle with stuff like eb or ct. (maybe even this splitted in multiple files)
- Logo-Client (which could be used 1:1 again perhaps?)
- Server stuff? Or just an example server to play with the examples?
- Partner stuff? (well..this may be a step for >2.0, since I don't know if any one of the typical suspects here has one/have ever dealed with such a thing .... also the main focus should be the Client anyway)

Just some suggestions from my side here.

gijzelaerr, 2024-07-04:
i fixed the examples as much as possible in https://github.com/gijzelaerr/python-snap7/pull/524. @swamper123 do you still plan to add these examples? otherwise we can close this issue.

closed 2024-07-05 by gijzelaerr