gijzelaerr/python-snap7 #272: Drop Snap7Exception

gijzelaerr, 2021-04-22:
I think we should rely less on `Snap7Exception`, and replace it with common Python exceptions.

For example, if a user supplies the wrong type to a function we should raise a `TypeError`.

swamper123, 2021-04-22:
I guess the only left Snap7Exception would be here:
https://github.com/gijzelaerr/python-snap7/blob/ee7ad61a4a3ec297a7ce31e9223255c01801ed68/snap7/common.py#L78

Since this is the main interface to get informations from the underneath snap7?

lautarodapin, 2021-04-22:
> I guess the only left Snap7Exception would be here:
> https://github.com/gijzelaerr/python-snap7/blob/ee7ad61a4a3ec297a7ce31e9223255c01801ed68/snap7/common.py#L78
> 
> Since this is the main interface to get informations from the underneath snap7?

Is there anyreason to keep using the informations from the snap7 lib? Having different exceptions in python will be nice for handling errors during development with python-snap7.

swamper123, 2021-04-22:
Our interface with snap7 is pretty small for thrown errors and dealing with thrown errors from snap7 in Python is horrible if it "just happens". It is not like a nice "ValueError" is thrown, in case stuff underneath exploded, they are much more ugly.
This check_error is a nice way to seperate our lib with the snap7 one. Also we are able to understand what was thrown from snap7 and could track down problems in an easy way.

Discussable would be creating new Exceptions/add messages to typical python ones like ValueError and use these...but I believe using builtin ones is enough for our usecases

gijzelaerr, 2021-04-22:
there are a couple
```
λ  grep -R "raise Snap7Exception" snap7
snap7/common.py:            raise Snap7Exception(msg)
snap7/common.py:        raise Snap7Exception(error)
snap7/client.py:            raise Snap7Exception(f"The cpu state ({state.value}) is invalid")
snap7/client.py:            raise Snap7Exception("The blocktype parameter was invalid")
snap7/client.py:            raise Snap7Exception("The blocktype parameter was invalid")
snap7/client.py:            raise Snap7Exception("The parameter was invalid")
snap7/client.py:            raise Snap7Exception("The parameter was invalid")
snap7/client.py:            raise Snap7Exception("The blocktype parameter was invalid")
snap7/util.py:                raise Snap7Exception("Max size could not be determinate. re.search() returned None")
snap7/util.py:                raise Snap7Exception("Max size could not be determinate. re.search() returned None")
snap7/partner.py:            raise Snap7Exception("The Client parameter was invalid")
snap7/logo.py:            raise Snap7Exception("The parameter was invalid")
snap7/logo.py:            raise Snap7Exception("The parameter was invalid")
```

gijzelaerr, 2021-04-22:
@lautarodapin i think we should keep all information available to us, that helps debugging problems. Just the class can be a bit more specific but also more Python std based.