gijzelaerr/python-snap7 #275: Use black?

gijzelaerr, 2021-04-22:
We could decide to use black.

How to proceed:
 * Run black on the source code, see if the changes are good and commit.
 * Add a github actions job that runs black, and fails if black changes the source code. That forces people to run black before committing.

swamper123, 2021-04-22:
Black should be configured to use XYZ characters, otherwise 88 is standard.

As much as I dislike commiting "remove space" things, I don't like "automated" stuff which is changing my code in a way, I may not want.

swamper123, 2021-07-06:
I have done some research on another project, but wanted to share this one:

https://wemake-python-stylegui.de/en/latest/

It seems that mypy in combination of wemake-python-styleguide (based on flake8) is a mighty tool to check the pythonstyle.

It is available as GitHubAction and may be a nicer way instead of using black and comparing codechanges.

mthuurne, 2022-01-19:
[`pre-commit`](https://pre-commit.com/) is a useful tool for this, since it can run Black automatically when Git creates commits, which saves time compared to finding out when CI alerts you a few minutes after pushing.

In CI, you can use `pre-commit run --all` to verify that Black and potentially other formatters and checkers were indeed run, without having to duplicate the configuration of which tool checks which sources.

We typically run Black, `isort`, `flake8` and `pre-commit`'s whitespace trimmer to eliminate most of the common formatting noise (`flake8` mainly to spot unused imports).

debauer, 2022-06-24:
You can also use Black in the Pipeline to enforce the correct Formatting. pre-commit is imho like checking password in the frontend.

mthuurne, 2022-06-24:
> You can also use Black in the Pipeline to enforce the correct Formatting. pre-commit is imho like checking password in the frontend.

You need to check in CI (pipeline) as well, which is why I recommended adding `pre-commit run --all` there, so the same configuration is applied both when committing and when checking pushes. If you're looking for an analogue, it's like validating form data both in the frontend and backend: frontend can point out errors to the user before the form is submitted while backend guards your data integrity.

debauer, 2022-06-24:
Okey. That's what i said. Enforce formatting with `black --check --diff`in your CI. How you made your local check/fixup is not relevant. You can use pre-commit hooks or a manually script. Until you don't enforce it in the CI, you can find a way around.

gijzelaerr, 2024-05-02:
we enabled the pre-commit

closed 2024-05-02 by gijzelaerr