gijzelaerr/python-snap7 #277: Merge logo code in client?

gijzelaerr, 2021-04-22:
the logo module was contributed a long time ago, and I always disliked the code duplication with snap7/client. maybe we can merge the module into the snap7/client code?

swamper123, 2021-04-22:
I haven't found logo stuff in snap7, so this is a duplicate. If the handling is a bit different than to other plcs, we should mention it in the docs.

lautarodapin, 2021-04-22:
> I haven't found logo stuff in snap7, so this is a duplicate. If the handling is a bit different than to other plcs, we should mention it in the docs.

The original lib doesn't have a specific module for the logo, but because the memory areas are diferent, it is a good thing to have them in another module.
Maybe we could add some inheritance for removing duplicated code?

swamper123, 2021-04-22:
@lautarodapin I am not so familiar with the logo ones. So there aren't DBs etc. or are they just rebranded?
We may need a table for that here to specify more, what has to be done.

lautarodapin, 2021-04-22:
> @lautarodapin I am not so familiar with the logo ones. So there aren't DBs etc. or are they just rebranded?
> We may need a table for that here to specify more, what has to be done.

This is a table from [this](https://www.npmjs.com/package/node-red-contrib-s7).

As you can see, we always use the same DB to referer every input, output, merkers, etc, and in different ranges we have the values.

Logo Block | Logo VM Range | example Node-RED address | Description
-- | -- | -- | --
I | 1024 - 1031 | DB1,BYTE1024 or DB1,X1024.5 or DB1,WORD1024 | Reads input terminals 1...8 or 6 or 1...16
AI | 1032 - 1063 | DB1,WORD1032 | Reads analog input terminal 1. Always word sized.
Q | 1064 - 1071 | DB1,BYTE1064 or DB1,X1064.5 or DB1,WORD1064 | Reads output terminals 1...8 or 6 or 1...16
AQ | 1072 - 1103 | DB1,WORD1072 | Reads analog output terminal 1. Always word sized.
M | 1104 - 1117 | DB1,BYTE1104 or DB1,X1104.5 or DB1,WORD1104 | Reads bit flags M1...M8 or M6 or M1...16
AM | 1118 - 1245 | DB1,WORD1118 | Reads analog flag 1. Always word sized.
NI | 1246 - 1061 | DB1,BYTE1246 or DB1,X1246.5 or DB1,WORD1246 | Reads network input 1...8 or 6 or 1...16
NAI | 1262 - 1389 | DB1,WORD1262 | Reads analog network input 1. Always word sized.
NQ | 1390 - 1405 | DB1,BYTE1390 or DB1,X1390.5 or DB1,WORD1390 | Reads network output 1...8 or 6 or 1...16
NAQ | 1406 - 1469 | DB1,WORD1406 | Reads network output 1. Always word sized.

This are the memory ranges.
### Logo 0BA7:
Type | Count | Start Address | Length
---- | ----- | ------------- | ------
Input | 24 | 923 | Bit
Analog Input | 8 | 926 | Word
Output | 16 | 942 | Bit
Analog Output | 2 | 944 | Word
Merker | 27 | 948 | Bit
Analog Merker | 16 | 952 | Word

### Logo 0BA8 / 8.SF4:
Type | Count | Start Address | Length
---- | ----- | ------------- | ------
Input | 24 | 1024 | Bit
Analog Input | 8 | 1032 | Word
Output | 20 | 1064 | Bit
Analog Output | 8 | 1072 | Word
Merker | 64 | 1104 | Bit
Analog Merker | 64 | 1118 | Word

openarchitect, 2024-05-29:
I wrote the logo stuff long time ago. I will check with the knowledge of today whether the separate logo file makes sense, or whether this can be merged into common by means of configuration. This has not the highest priority - this can take some weeks.

gijzelaerr, 2024-07-04:
@openarchitect did you manage to compare the two implementations?

I'm preparing a 2.0 release at the moment, so it would be nice to include these changes or we close this ticket.

gijzelaerr, 2024-07-05:
i've change the class so it inherrits from the Client class and removed the duplicate functionality. The only difference is now the way the connection is done and there are Logo specific read/write functions. I think this is good enough for now.

closed 2024-07-05 by gijzelaerr