gijzelaerr/python-snap7 #280: Function Refused By CPU

kevilay, 2021-04-23:
Having an issue with try to read a DB or memory Area I will list below two sample codes I tried. My connection is successful but I get the rejection on the read function. In the memory example I am trying to read M0.0. For the DB example I have created a global DB200 with only 2 bits inside it and I am trying to read DB200.00. The db optimization is turned off as per the instruction.

CPU is 6ES7212-1AE40-0XV0 CPU 1212-C - DC/DC/DC Firmware Version 4.4

Reading Memory Example:

import snap7.client as c
from snap7.util import *
from snap7.types import *
import socket

```
def ReadMemory(plc,byte,bit,datatype):
    result = plc.read_area(areas['MK'],0,byte,datatype)
    if datatype == S7WLBit:
        return get_bool(result,0,1)
    elif datatype == S7WLByte or datatype == S7WLWord:
        return get_int(result,0)
    elif datatype == S7WLReal:
        return get_real(result,0)
    elif datatype == S7WLDWord:
        return get_dword(result,0)
    else:
        return None

if __name__ == "__main__":
    plc = c.Client()
    plc.connect('192.168.1.10',0,1)
    print("Connection Sucessful") 
    print(ReadMemory(plc,0,0,S7WLBit))
```

@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@

```
Example two Reading DB

import snap7.client as c
from snap7.util import *
from snap7.types import *
import socket

def ReadMemory(plc,byte,bit,datatype):
    result = plc.read_area(area = areas['DB'],dbnumber = 200,start = 0,size = 0x01)
    if datatype == S7WLBit:
        return get_bool(result,0,1)
    elif datatype == S7WLByte or datatype == S7WLWord:
        return get_int(result,0)
    elif datatype == S7WLReal:
        return get_real(result,0)
    elif datatype == S7WLDWord:
        return get_dword(result,0)
    else:
        return None

if __name__ == "__main__":
    plc = c.Client()
    plc.connect('192.168.1.10',0,1)
    print("Connection Sucessful") 
    
    print(ReadMemory(plc,0,0,S7WLBit))
```

Both cases are giving me the same error. I am a pretty experienced siemens programmer but fairly new at Python. Please let me know if you see any errors or what you suggest?

kevilay, 2021-04-24:
Iautarodapin that was exactly the problem! Thank you very much for your help.

lautarodapin, 2021-04-24:
> Iautarodapin that was exactly the problem! Thank you very much for your help.

Your welcome!

FabioAlmeida25, 2021-06-23:
Hello I have the same problem, how I do that on TIA Portal ?

swamper123, 2021-06-28:
@FabioAlmeida25 Please open a new Issue with some backround information (setup? what have you tried? etc....)