gijzelaerr/python-snap7 #289: No module found

INDRANI125, 2021-06-18:
I was trying to install Snap7 Library on Windows 2012 Server R2 OS where I have installed latest python version and downloaded Snap7 from link provided in documentation for Windows and also have copied Snap7 DLL file in Windows->System32 folder as well but while running python program, I am getting No module found error. Can any one help? or provide me guide what I am doing wrong while installing?

gijzelaerr, 2021-06-18:
can you try to install the binary wheel for windows from here:

https://github.com/gijzelaerr/python-snap7/suites/2822028160/artifacts/62930376

this is the new version which we will release at some point and has the snap7 dll build in.

INDRANI125, 2021-06-21:
How to install python_snap7-1.0-py3-none-win_amd64 present inside windows folder of wheels?

INDRANI125, 2021-06-21:
Running Now.Thank u :-)

closed 2021-06-21 by INDRANI125

gijzelaerr, 2021-06-21:
So the windows binary wheel is working for you?

I think it is time I upload them to pypi then :)

INDRANI125, 2021-06-22:
Yes :)