gijzelaerr/python-snap7 #293: client.db_read() command blocking

Dhruva0707, 2021-06-28:
Hello I have a S7-1200 PLC, a code snippet is as follows:

swamper123, 2021-06-28:
@Dhruva0707 I guess there is something missing ;)

Dhruva0707, 2021-06-28:
> @Dhruva0707 I guess there is something missing ;)

Updated

swamper123, 2021-06-28:
1) `db_read ` is not async, so it is blocking. 
2) You may want to use `as_db_read` and waiting until a result exists with `wait_as_completion` or checking if now a result exist with `check_as_completion` (last one can be used in a loop).

Dhruva0707, 2021-06-28:
Thanks a lot for your response!

I will check it out. 

As I am reading the documentation I see that a 'buffer' is required in the as_db_read function. Any tutorial or example code I see to get the datatype of the buffer and using 'wait_as_completion' or 'check_as_completion' ?

Dhruva0707, 2021-06-28:
Found it in the tests

Dhruva0707, 2021-06-30:
We can close this issue, as the async call resolved the same.

closed 2021-06-30 by Dhruva0707