gijzelaerr/python-snap7 #339: bug introduced in version 1.1

2makeitwork, 2022-01-16:
In version 1.1, following exception is thrown when executing "client.read_area(131, 0 , 1, 1)"
TypeError("unsupported operand type(s) for 'in': 'int' and 'EnumMeta'"
The same command runs without issue in version 0.11.

By step in debug, it is found the newly introduced when testing whether or not an integer is in enumeration s7type.Areas causes the exception in client.py.
The enum Areas did not exist in version 0.11.

swamper123, 2022-01-17:
Could you replace `131` with `snap7.types.Areas.MK` ?

Enums handle stuff a bit different than the previous version, but imo this is the better way. It is more readable.

spreeker, 2022-01-17:
Breaking existing solutions is *_very_* dangerous..  "Oh now I changed the
code to look better", and btw oops exploded your equipment...

Senior Software Engineer.
+31 6 24 69 22 97
https://gitlab.com/spreeker/
https://github.com/spreeker/
https://www.nlx.io/
https://commondatafactory.nl/

2makeitwork, 2022-01-17:
I agree with the points from both of you. The change is dangerous AND improves readability.
There are better ways to do this, true, but nobody paid anything for it, right? 
So, relax, thank the author.
If the equipment does 'explodes', that's even better. No problem, no budget.

swamper123, 2022-01-17:
@spreeker This is true. I still would expect breaking changes in a major release from 0.X.Y to 1.X.Y. (Best example: Python itself).
Anyway, we should have mentioned them in the Changelogs anywhere, indeed.
I have no opinion about pro or con `Areas(Enum)`, but checking if that parameter is allowed is still important imo.
Atm I haven't got a good alternative (only list or a dict maybe?) to catch all cases.

gijzelaerr, 2022-01-18:
the `0.x` versioning sort of implies a pre-release (although it took many years to get to `1.x`), so we gave ourselves the liberty to change the API here and there. If things break for you you can still use the older version. I think we introduced this change for the 1.0 release since using an actual type is less error-prone, you can now not pass on an invalid area int value anymore.

We could support both types `Areas | int`, but I think it is just best to stick with the type implementation we came up with.

I consider your question answered and this issue not really a bug, just a change in the final API. Feel free to reopen if you disagree.

closed 2022-01-18 by gijzelaerr

nikteliy, 2022-01-18:
@2makeitwork They're right, but I'm on your side.
@gijzelaerr I made a pull request #343 in case you change your mind.

mthuurne, 2022-01-19:
Note that this change was first released in 1.1; version 1.0 didn't have the `Areas` enum yet.

While I agree that an enum is a nicer interface than an integer, it would be good to still accept integers for a few releases and print a deprecation warning when they're used. The type annotation could allow `Areas` only to push people towards the new interface.