gijzelaerr/python-snap7 #344: how to read out package version

2makeitwork, 2022-01-20:
As things are different in ver 0.11 and 1.1, I need to get the version of installed package to adapt this 2 versions.
the snap7.__version__ seems to be always 0.0rc0 regardless of the installed package version.
Or do I get it wrong somehow?

(base) super@daq:~$ pip install python-snap7==1.1
Collecting python-snap7==1.1
  Using cached python_snap7-1.1-py3-none-manylinux1_x86_64.whl (132 kB)
Installing collected packages: python-snap7
  Attempting uninstall: python-snap7
    Found existing installation: python-snap7 0.11
    Uninstalling python-snap7-0.11:
      Successfully uninstalled python-snap7-0.11
Successfully installed python-snap7-1.1
(base) super@daq:~$ python
Python 3.9.7 (default, Sep 16 2021, 13:09:58) 
[GCC 7.5.0] :: Anaconda, Inc. on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import snap7
>>> snap7.__version__
'0.0rc0'

>>> 
(base) super@daq:~$ pip install python-snap7==0.11
Collecting python-snap7==0.11
  Using cached python_snap7-0.11-py3-none-any.whl
Installing collected packages: python-snap7
  Attempting uninstall: python-snap7
    Found existing installation: python-snap7 1.1
    Uninstalling python-snap7-1.1:
      Successfully uninstalled python-snap7-1.1
Successfully installed python-snap7-0.11
(base) super@daq:~$ python
Python 3.9.7 (default, Sep 16 2021, 13:09:58) 
[GCC 7.5.0] :: Anaconda, Inc. on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import snap7
>>> snap7.__version__
'0.0rc0'
>>>

gijzelaerr, 2022-01-20:
> (base) ***@***.***:$ pip install python-snap7==1.1
> (base) ***@***.***:$ python
> (base) ***@***.***:$ pip install python-snap7==0.11

That’s a bug :)

gijzelaerr, 2022-01-20:
and confirmed:
```
$  python3 -m venv venv
$  venv/bin/pip install python-snap7
Collecting python-snap7
  Using cached python_snap7-1.1-py3-none-macosx_10_9_universal2.whl (190 kB)
Installing collected packages: python-snap7
Successfully installed python-snap7-1.1
gijs@bloempot-4:~/tmp
$  venv/bin/python -c "import snap7; print(snap7.__version__)"
0.0rc0
```

closed 2022-01-20 by gijzelaerr