gijzelaerr/python-snap7 #353: errormessage b'CPU : Item not available' when using get_cpu_info()

smar2022, 2022-03-02:
Hello,

I allwaysreceive the following errormessage:

b'CPU : Item not available'
Traceback (most recent call last):
  File "C:\Users\smar\Python\Projects\CMAD\firsttry.py", line 169, in <module>
    main()
  File "C:\Users\smar\Python\Projects\CMAD\firsttry.py", line 126, in main
    info = s7.get_cpu_info()
  File "C:\Users\smar\AppData\Roaming\Python\Python310\site-packages\snap7\client.py", line 161, in get_cpu_info
    check_error(result, context="client")
  File "C:\Users\smar\AppData\Roaming\Python\Python310\site-packages\snap7\common.py", line 92, in check_error
    raise Snap7Exception(error)
snap7.exceptions.Snap7Exception: b'CPU : Item not available'

 when using the get_cpu_info function. All settings are right (RACK = 0, SLOT = 1, Put/Get access allowed, etc.)
I use a S7 1200 FW 4.2 and TIA Portal V17
Can anyone help?

swamper123, 2022-03-02:
Since we receive a legit Error from the CPU, it may be the Firmware or a HW config still not set well...

Normal stuff like reading variables works?

closed 2022-03-02 by smar2022

swamper123, 2022-03-02:
Just see #323 had a similar Problem (and I haven't answered :sweat_smile: ). 
What I know is, that the 1200 and 1500 are not supporting all features.

I am sorry, but atm I haven't got a device on hand to verify this.

Since this doesn't seem to be an Python issue, you may ask at the [snap7 directly.](https://sourceforge.net/p/snap7/discussion/general/). Davide is very helpfull and just use his great work as a python-wrapper. :)

smar2022, 2022-03-02:
OK, THX.