gijzelaerr/python-snap7 #372: Use correct typing

debauer, 2022-06-24:
You use typing -> Great
You return the wrong Type -> Not so Great
You use strange typing -> okey but not Great

Examples:
wrong use: util get_byte returns int, is typed as bytes
strange use: util set_byte wants int 
no use: some return types and some arguments are not typed

Possible Solution:
enforce mypy 

Better possible Solution:
use poetry, pylint, black (or similar tools) and enforce it in your pipeline 

if you want i can participate and create a PR, but it can take some time.

swamper123, 2022-06-24:
Thanks for the hint.
PRs are always welcomed :)

gijzelaerr, 2022-06-24:
we already use mypy and flake8:

https://github.com/gijzelaerr/python-snap7/tree/master/.github/workflows

debauer, 2022-06-24:
Okey. I'm to biased from my default config. 
default mypy is to lazy and imho useless. It enforce nothing. It only checks what you have typed. Not what you not have typed.
Enforce it and you will never have typing problems at the runtime. Than you can't return an int if you typed the return as bytes. 

My pyproject.toml contains: 

```warn_return_any = true
warn_unused_configs = true
warn_unused_ignores = true
warn_redundant_casts = true
warn_no_return = true

no_implicit_optional = true
warn_unreachable = true
disallow_untyped_defs = true
disallow_incomplete_defs = true
# disallow_any_explicit = true
disallow_any_generics = true
disallow_untyped_calls = true
```

for nicer output: 

```
# output style configuration
show_column_numbers = true
show_error_codes = true
pretty = true
``

gijzelaerr, 2024-05-02:
i think we fixed the typing issues.

closed 2024-05-02 by gijzelaerr