gijzelaerr/python-snap7 #374: RuntimeError: "can't find snap7 library" on Raspbian GNU/Linux 10 (buster)

ts4iot, 2022-06-30:
I get the following error on a fresh installed Raspian buster:

```
_Python 3.7.3 (default, Jan 22 2021, 20:04:44)
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import snap7
>>> g_plc = snap7.client.Client()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/deviceadmin/.local/lib/python3.7/site-packages/snap7/client.py", line 64, in __init__
    self._library = load_library(lib_location)
  File "/home/deviceadmin/.local/lib/python3.7/site-packages/snap7/common.py", line 72, in load_library
    return Snap7Library(lib_location).cdll
  File "/home/deviceadmin/.local/lib/python3.7/site-packages/snap7/common.py", line 63, in __init__
    raise RuntimeError("can't find snap7 library. If installed, try running ldconfig")
RuntimeError: can't find snap7 library. If installed, try running ldconfig_
```

python-snap7 is successfully installed with: sudo pip3 install python-snap7

```
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting python-snap7
  Downloading https://www.piwheels.org/simple/python-snap7/python_snap7-1.2-py3-none-any.whl
Installing collected packages: python-snap7
Successfully installed python-snap7-1.2
```

Is it possible to install the missing library?

ts4iot, 2022-06-30:
Yes, but without sucess.

swamper123, 2022-06-30:
pyton-snap7 is a wrapper around the library of snap7.
Normally it got installed within the wheel process as well (including the ldconfig part). So not sure what happened here.
In such cases a manual install has to be done.

gijzelaerr, 2022-07-04:
does `dpkg --print-architecture` give `armhf`? Wheels are not supported for ARM installations, so it falls back on a source install, which doesn't include the .so file. You need to manually install the .so file, which you somehow also did on your stretch machine.

gijzelaerr, 2022-07-04:
I also noticed you are using wheels from https://www.piwheels.org/, which are not made by us and not supported by us. It seems this is a 3rd party service that builds our wheels without including the so.

ts4iot, 2022-07-04:
Yes I can confirm. dpkg --print-architecture gives armhf and getconf LONG_BIT returns 32. 
I cannot remember to install it on Stretch, but it was long time ago.

ts4iot, 2022-07-04:
It is a Raspberrypi based hardware platform. 

https://https://revolutionpi.de/

cmoutafidis, 2023-09-18:
@gijzelaerr can tou provide further information on how to manually install the .so file?

nikteliy, 2023-10-05:
@cmoutafidis @ts4iot 
You could try installing [this package](https://github.com/gijzelaerr/python-snap7/suites/15112242354/artifacts/863571533). It's built for the armv7l arch.

@gijzelaerr I think we haven't added armv7l package to PyPI yet

gijzelaerr, 2023-10-05:
thanks @nikteliy, good point. i've uploaded the wheel, 1.3 should now be installable on `armv7l`. @cmoutafidis can you verify?