gijzelaerr/python-snap7 #379: In the DB class, there is no canonical way to iterate on all rows when id_field is not None

vmsh0, 2022-08-01:
When `id_field` is set to something that is not `None`, there is no canonical way or public API to iterate on all rows or even to obtain a list of the available keys.

The only way to do that currently is by accessing the `index` field, which however is not documented and supposedly can't be presumed to be part of the public API. (Furthermore, even if it was, directly accessing attributes to get data is usually bad API design, although the `@property' decorator somewhat mitigates the associated problems.)

What I propose is one or more of - with my personal preference being the first two items of the list:
- making `DB` iterable like dictionaries
- having a `DB.items()` method like that of dictionaries
- having a `DB.export()` method like that of `DB_Row`

Like for the other issues, I'm available to submit PRs. However, since I already have two PRs pending, I'm going to wait for your feedback on those and this issue before opening more.

vmsh0, 2022-08-22:
I just noticed the class has an `__iter__` method without any documented semantics. It iterates over the values. This method, in other mappable types such as `dict`, iterates over the _keys_, not the values.

Since I'm guessing it's too late to change that interface, I'm going to put forward a PR soon with the following changes:
- `keys()` and `__contains__()` methods, with `dict` semantics
- `export()` method, flattening all available data to a plain `dict`

spreeker, 2022-08-22:
Hai that might me a good idea. At the time I was in a hurry building all of this.

I do remember there where some random inconsistency between DB objects with different starting offsets. So I let go of solving that In the abstract class.

I've implemented code that every few seconds would read the entire DB and did updates to some values. The update would take 5ms. On a single value. The reading of the entire DB could take up almost a whole second and was a rather hefty operation (for a plc). Doing a key by key update is slowish.

It is as fast as possible.

My fist Implantation involved some OPC middle ware and took seconds and was unrealizable when I was reading just a few variables.

closed 2022-11-08 by gijzelaerr