gijzelaerr/python-snap7 #380: In the DB class, there is no canonical way to read or write the entire DB

vmsh0, 2022-08-01:
When using the `DB` class, you provide initial data, and then you're able to provide a new underying `bytearray` to refresh the data.

While `DB_Row` allows to write to and read from the PLC to move data in and out of this bytearray (each row on its own slice), there is no way to write the entire `DB` (i.e. all the rows) at once. This is especially detrimental to our use case because we have a large number of rows, and separating the reads/writes is degrading our application performance.

As far as I can make out, when `row_offset` is 0 there is no real problem with reading and writing all the rows at once. As such, I propose the implementation of the `DB.read(client)` and `DB.write(client)` methods.

Again, I'm available & happy to submit PRs after any relevant feedback on the existing PRs and issues.

spreeker, 2022-08-06:
This is implemented.

https://www.youtube.com/watch?v=G-Gj_r2BQBk

vmsh0, 2022-08-08:
Hi,

> This is implemented. Rather Dangerous Operation and you better be sure the mapping is correct :).
> 

I absolutely agree it's a dangerous operation, just as dangerous as writing any value at all on a live system. This is why we have the `DB` class: you define your mapping and then the class sticks to it instead of you having to do that. It's an abstract model. So I argue that no, it is not implemented: currently, to perform the operation, you need to leave the abstract model and manipulate a byte buffer and some loosely-linked metadata.

Other than the purely pragmatic argument that it makes the model more expressive, it's also a matter of coherence: why is it possible to write back single rows, but not all of them? (in some cases that's _literally_ not possible, when using the `DB` abstraction, without working with non-public parts of the API or - again - manually manipulating the byte buffer and loosely-linked metadata; see #379)

> https://www.youtube.com/watch?v=G-Gj_r2BQBk

Sorry, I'm not sure what I'm supposed to be seeing in this video!

> 
> def write_data_db(dbnumber, all_data, size):

vmsh0, 2022-08-22:
Just for sake of reference, I have now documented in #379 that there is a way after all to iterate on keys, with some caveats.

closed 2022-11-08 by gijzelaerr