gijzelaerr/python-snap7 #407: implement `__main__`

Cube707, 2022-12-06:
It would be helpful and more pythonic to implement a `__main__.py`. This would allow to start the server like in many other packages using:

```bash
python -m snap7
```

even better, by implementing _server_ and _client_ a proper sub-packages of _ snap7_ it would allow for the even more verbose syntax: (however, this might break the current api and might requirte a major version change)

```bash
python -m snap7.server
python -m snap7.client
```  

Please assigne me this issue if it is of interest to you and I should implement it

gijzelaerr, 2022-12-07:
i'm open to this proposal, but I prefer not to introduce API changes. how would you supply the arguments to starting the client that way?

Cube707, 2022-12-07:
I have tested this a little and it seems to work without requrirering API changes (will investigate further)

Suppying arguments is verry easy, just slap them on the end. For example, just moving `bin/snap7-server.py ->server2/__main__.py` is enough to use this: (the `2` is jsut to avoid naming collisions)

```python
python -m snap7.server2 path/to/snap7.dll
```

see [the python docs](https://docs.python.org/3/library/http.server.html#http-server-cli)  on how the default `http.server` can be used. This is what I am talking ybout

If you are interesed I can open a PR for further discussion.

gijzelaerr, 2022-12-07:
Sure, a PR is always welcome. I would not make the path to the DLL a function argument though since it is already configurable by setting the `PATH` environment variable and in general not required to be supplied.

Cube707, 2022-12-07:
> I would not make the path to the DLL a function argument

Than I missunderstood what the current implementation is doing with its first argument. I agree that this should definitely not be a required argument (maybe an optional though?). It was just an example...

PR #409 (2022-12-20): implement `__main__.py` enty points
This PR implements entripoints via `__main__.py` files, which closes #407. This can replace the existing `bin/snap7-server.py` script and offer a much more pythonic way to achive the same.

Currently implemented:

``` bash
Usage: python -m snap7.server [OPTIONS]

  Command entrypoint to start a dummy server.

Options:
  -p, --port INTEGER  Port the server will listen on.
  --dll FILE          Path to the snap7 DLL (if it is not on path).
  --help              Show this message and exit.
```

### Diskussion

There are still a few question I feel should be dis [... cut]