gijzelaerr/python-snap7 #413: using as_write_area Not able to write a single bit

Diptee24, 2022-12-29:
hi sir,
i am getting below error for snap7 as_write_area function.
Traceback (most recent call last):
  File ".\forumsend.py", line 17, in <module>
    client.as_write_area(snap7.types.S7AreaDB,1,56,1,snap7.types.S7WLBit,byte_array_true)
  File "....\lib\site-packages\snap7\client.py", line 1053, in as_write_area
    logger.debug(f"writing area: {area.name} dbnumber: {dbnumber} start: {start}: size {size}: "
AttributeError: 'int' object has no attribute 'name'

my code is:
import sys
import snap7
from snap7.types import areas,Areas
from ctypes import byref

ANPIP='192.168.2.1'

byte_array_true = bytearray([1])
byte_array_false = bytearray([0])

client = snap7.client.Client()
client.set_connection_type(3)
client.connect(ANPIP, 0, 1)

buffer1 = client.db_read(1, 0, 5000)

client.as_write_area(snap7.types.S7AreaDB,1,56,1,snap7.types.S7WLBit,byte_array_true)

gijzelaerr, 2022-12-29:
you should supply `snap7.types.Areas.DB` to that function, the `S7Area*` constants are deprecated.

closed 2022-12-29 by gijzelaerr

Diptee24, 2022-12-30:
still getting same error

Traceback (most recent call last):
  File ".\forumsend.py", line 17, in <module>
    client.as_write_area(snap7.types.Areas.DB,1,56,1,snap7.types.S7WLBit,byte_array_true)
  File "\lib\site-packages\snap7\client.py", line 1056, in as_write_area       
    res = self._library.Cli_AsWriteArea(self._pointer, area.value, dbnumber, start, size, wordlen.value, byref(cdata))
AttributeError: 'int' object has no attribute 'value'

import sys
import snap7
from snap7.types import areas,Areas
from ctypes import byref

ANPIP='192.168.2.1'

byte_array_true = bytearray([1])
byte_array_false = bytearray([0])

client = snap7.client.Client()
client.set_connection_type(3)
client.connect(ANPIP, 0, 1)

buffer1 = client.db_read(1, 0, 5000)

client.as_write_area(snap7.types.Areas.DB,1,56,1,snap7.types.S7WLBit,byte_array_true)

gijzelaerr, 2022-12-30:
now it is the wordlen argument, should be `snap7.types.WordLen.Bit`

Diptee24, 2022-12-30:
okay i have changed the same...error is solved...thank you so much for solution and quick response.