gijzelaerr/python-snap7 #420: debian executable issue due to snap7

Diptee24, 2023-01-24:
i made executable using pyinstaller and my application use snap 7 library.
after generating executable successfull when run executable it give snap7 error.

gijzelaerr, 2023-01-24:
What is the error

Diptee24, 2023-01-24:
can't find snap7 library. If installed, try running ldconfig
but i have already installed using sudo python -m pip install python-snap7 and i can run using terminal

spreeker, 2023-01-24:
python-snap7 is a wrapper around a C library called snap7. that C library is apparently missing. read documentation.

closed 2023-01-24 by spreeker

Diptee24, 2023-01-24:
but it works from terminal and from where i can get this file and where i have to put

gijzelaerr, 2023-01-24:
you need to figure out how to bundle the snap7 shared libraries into your pyinstaller package, that is out of scope for python-snap7.