gijzelaerr/python-snap7 #445: get_string NOT working as described in docs

Picca-SNI, 2023-04-11:
Trying to read a string from a S7-1500 PLC, documentation says util.get_string takes 3 arguments (bytearray, index, max_size) however, using the function like this does not work, because it only takes two arguments (bytearray, index) 

terminal output:
Ext_PO_number = get_string(dataBuffer, EXT_PO_NUM_START, EXT_PO_NUM_LEN)
TypeError: get_string() takes 2 positional arguments but 3 were given
Command exited with return code 1
ERROR - Task failed with exception

nikteliy, 2023-04-11:
Hi @Picca-SNI 
Could you provide a link to the documentation with the error?
According to the [official documentation](https://python-snap7.readthedocs.io/en/latest/API/util.html#snap7.util.get_string), the `get_string()` function only takes two arguments.

Picca-SNI, 2023-04-11:
My bad ... apparently google sent me to the docs for version 1.0 ....

closed 2023-04-11 by Picca-SNI