gijzelaerr/python-snap7 #454: Function Refused By CPU

GleeisonRodrigues, 2023-08-03:
Having an issue with try to read a DB. My connection is successful but I get the rejection on the read function. In the DB example I have created a global DB26 and I am trying to read DB26.00. The db optimization is turned off as per the instruction and also I enable the features ALLOW PUT/GET .

CPU is 6ES7 214-1HG40-0XB0 CPU 1214C DC/DC/Rly Firmware Version 4.2

python version 3.11.4

Reading Example:

import snap7.client as c
from snap7.util import *
from snap7.types import *

if __name__ == "__main__":
    plc = c.Client()
    plc.connect('192.168.0.20',0,1)
    print("Connection Sucessful")
    data = plc.db_read(26,0,1)

[image]

[image]

[image]

nikteliy, 2023-08-03:
Hi @GleeisonRodrigues
Check the properties of your DB block. The `optimized block access` checkbox should be unchecked.

[image]

[image]

GleeisonRodrigues, 2023-08-03:
Hi @nikteliy , I did that.
[image]

nikteliy, 2023-08-03:
Did you reinitialize the db after that? It works for me, but i have firmware 4.0.
[image]

GleeisonRodrigues, 2023-08-03:
Hi I also did that :(