gijzelaerr/python-snap7 #463: snap7.util.DB_Row  set_dtl

gunaycomert, 2023-08-21:
when added "set_dtl" funtion in util

gunaycomert, 2023-08-21:
def set_dtl(bytearray_: bytearray, byte_index: int, dt: datetime) -> bytearray:
    bytearray_[0:2] =  dt.year.to_bytes(2, byteorder='big', signed=True)
    bytearray_[2:3] =  dt.month.to_bytes(1, byteorder='big', signed=True)
    bytearray_[3:4] =  dt.day.to_bytes(1, byteorder='big', signed=True)
    bytearray_[4:5] =  dt.isocalendar()[1].to_bytes(1, byteorder='big', signed=True)
    bytearray_[5:6] =  dt.hour.to_bytes(1, byteorder='big', signed=True)
    bytearray_[6:7] =  dt.minute.to_bytes(1, byteorder='big', signed=True)
    bytearray_[7:8] =  dt.second.to_bytes(1, byteorder='big', signed=True)
    bytearray_[8:12] =  dt.microsecond.to_bytes(4, byteorder='big', signed=True)
    return bytearray_

gijzelaerr, 2023-10-10:
is this a bug?

closed 2023-10-10 by gijzelaerr