gijzelaerr/python-snap7 #465: Too much logging in client db_read() method

jwhitmor, 2023-08-31:
I've see issue: https://github.com/gijzelaerr/python-snap7/issues/238, but it would be advantageous to be able to switch off the logging message in client db_read() method.

When you're reading back 100s of datapoints it becomes noise. The logging level would impact other useful logging of establishing connections etc. So to keep that logging info, but curtail the db_read() logging. It would be good if a parameter "log=True" could be added to the method, so that no existing logging is impacted. As the default would be true no existing code is impacted, but when required logging on a db_read() call could be set to false with log=False.

gijzelaerr, 2023-08-31:
the reason why we use the logging library with log levels is to introduce this kind of code. The logging in the `db_read()` function is already set to debug. The purpose of the debug level is to debug, not to run and log in a production environment. are you saying you need to run the debug level in production? This could mean a higher level function calling `db_read()` should have a higher logging level and you should only listen for INFO and higher.

gijzelaerr, 2023-10-10:
closing this issue since no more activity in the discussion.