gijzelaerr/python-snap7 #474: Add python 3.12 support, build wheels

gijzelaerr, 2023-10-24:
we can also drop python 3.7 support since it is EOL.

nikteliy, 2023-11-01:
I tried to build a package for Python 3.12 and this is what I found:
- it doesn't work with [old setuptools](https://stackoverflow.com/questions/77364550/attributeerror-module-pkgutil-has-no-attribute-impimporter-did-you-mean)
- only new versions of setuptools build with pyproject.toml ( >=64)
- new versions of setuptools doesn't handle "--build-option" parameters correctly
[image]

Therefore, we cannot use either new or old versions of setuptools. I suggest we just wait until setuptools is fixed

gijzelaerr, 2024-05-02:
support is there now, we only need to fix the build :)

closed 2024-05-03 by gijzelaerr