gijzelaerr/python-snap7 #479: Error in util.Set_String and util.Get_String

PoitrasJ, 2023-12-06:
When reading the first Byte in the ByteArray, which contains the maximum length of the string, the Hexadecimal value is sometimes in a format that Python will not cast to Integer : b'\ff' will stay the type "Bytes" and the slice access fails.

this Section in set_string
```
    # fill the rest with empty space
    for r in range(i + 1, (bytearray_[byte_index]) - 2):
        bytearray_[byte_index + 2 + r] = ord(' ')
```
this Section in get_string
```
    str_length = int(bytearray_[byte_index + 1])
    max_string_size = int(bytearray_[byte_index])
```
The solution i came up is to evaluate the value using ord  :

```
    # fill the rest with empty space
    for r in range(i + 1, ord(bytearray_[byte_index]) - 2):
        bytearray_[byte_index + 2 + r] = ord(' ')
```
```
    str_length = ord(bytearray_[byte_index + 1])
    max_string_size = ord(bytearray_[byte_index])
```

gijzelaerr, 2024-07-04:
Can you supply a unit test or string to break the current behaviour?

lupaulus, 2024-11-09:
[image]

@gijzelaerr The solution proposed is not working