gijzelaerr/python-snap7 #486: db[0]['x']=22.33 raises ValueError

thkfighter, 2024-01-31:
```Python
import snap7

config = {
    "user_name": "admin",
    "password": "admin",
    "locator_host": "127.0.0.1",
    "locator_pose_port": 9011,
    "locator_json_rpc_port": 8080,
    "plc_host": "192.168.8.71",
    "plc_port": 102,
    "plc_rack": 0,
    "plc_slot": 1,
    "seed_count": 8,
    "db_number": 10000,
    "row_size": 26,
    "pose_size": 24,
    "layout": """
0       x               LREAL
8       y               LREAL
16      yaw             LREAL
24.0    enforceSeed     BOOL
24.1    uncertainSeed   BOOL
24.2    recordSeed      BOOL
24.3    setSeed         BOOL
""",
    "debug": 0,
}

client = snap7.client.Client()
client.connect(
    config["plc_host"], config["plc_rack"], config["plc_slot"], config["plc_port"]
)
all_data = client.db_read(
    config["db_number"], 0, config["row_size"] * config["seed_count"]
)
# all_data = client.upload(config["db_number"])
db = snap7.util.DB(
    db_number=config["db_number"],
    bytearray_=all_data,
    specification=config["layout"],
    row_size=config["row_size"],
    size=config["seed_count"],
)
db[0]['x']=22.33
```

```Bash
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[8], line 1----> 1 db[0]['x']=22.33

File c:\workspace\locator_plc_bridge\venv\Lib\site-packages\snap7\util.py:1657, in DB_Row.__setitem__(self, key, value)
   1655 def __setitem__(self, key, value):
   1656     index, _type = self._specification[key]
-> 1657     self.set_value(index, _type, value)

File c:\workspace\locator_plc_bridge\venv\Lib\site-packages\snap7\util.py:1823, in DB_Row.set_value(self, byte_index, type_, value)
   1820 if type_ == 'TIME' and isinstance(value, str):
   1821     return set_time(bytearray_, byte_index, value)
-> 1823 raise ValueError

ValueError: 
```

spreeker, 2024-01-31:
https://github.com/gijzelaerr/python-snap7/blob/master/snap7/util.py#L1806   You are right!! the LREAL type is missing here!

spreeker, 2024-01-31:
https://github.com/gijzelaerr/python-snap7/pull/487

closed 2024-05-01 by gijzelaerr