gijzelaerr/python-snap7 #493: Improve doc for area_read in client

bruceunx, 2024-03-26:
```python  
  def read_area(self, area: Areas, dbnumber: int, start: int, size: int) -> bytearray:
        """Reads a data area from a PLC
        With it you can read DB, Inputs, Outputs, Merkers, Timers and Counters.

        Args:
            area: area to be read from.
            dbnumber: number of the db to be read from. In case of Inputs, Marks or Outputs, this should be equal to 0.
            start: byte index to start reading.
            size: number of bytes to read.

        Returns:
            Buffer with the data read.

        Raises:
            :obj:`ValueError`: if the area is not defined in the `Areas`

        Example:
            >>> import snap7
            >>> client = snap7.client.Client()
            >>> client.connect("192.168.0.1", 0, 0)
            >>> buffer = client.read_area(Areas.DB, 1, 10, 4)  # Reads the DB number 1 from the byte 10 to the byte 14.
            >>> buffer
            bytearray(b'\\x00\\x00')
        """
```
> dbnumber: number of the db to be read from. In case of Inputs, Marks or Outputs, this should be equal to 0.

from snap7 reference 
> DB Number if Area = S7AreaDB, otherwise is ignored.

I think maybe use `dbnumber: number of the db to be read from when area type is DB, otherwise is ignored` to consist with snap7 and less confused.

lupaulus, 2024-03-29:
I Agree, the documentation need better reference related to the snap7 library 

The set "calls this should be equal to 0", have not the same signification then "otherwise is ignored"

gijzelaerr, 2024-05-02:
we happily accept PRs with improvements.

gijzelaerr, 2024-07-04:
implemented in https://github.com/gijzelaerr/python-snap7/pull/524

closed 2024-07-05 by gijzelaerr