gijzelaerr/python-snap7 #517: version 1.4 broken on python 3.8

haronaut, 2024-05-14:
Tested on python 3.8.10

If I run `import snap7` I get the following error:

`site-packages/snap7/server/__init__.py", line 77, in Server
    def register_area(self, area_code: int, index: int, userdata: ctypes.Array[ctypes.c_int8]):
TypeError: '_ctypes.PyCArrayType' object is not subscriptable`

python-snap7 version 1.3 does not have this issue.

gijzelaerr, 2024-05-14:
duplicate of https://github.com/gijzelaerr/python-snap7/issues/511

i think i should retract 1.4 since that release  was incorrectly tagged as supporting python 3.7

closed 2024-05-14 by gijzelaerr

gijzelaerr, 2024-05-14:
`1.4` has been "yanked". `Python > 3.8` should now resolve to `python-snap7=>1.4.1`, `python=<3.8` to `python-snap7<1.4`