gijzelaerr/python-snap7 #521: Add set_pdu_length()

luewh, 2024-06-12:
The default PDU length during negociation is 480, but s1500 PLC allows up to 960, is would be good to enable such setting.

I can to it without modifying the client.Client class by setting the PDURequest param before performe a connection.

```
clientInstance = client.Client()
clientInstance.set_param(10, pduSize)
clientInstance.connect("127.0.0.1", 0, 0, 102)
```
But I think It would be more direct by passing the pduSize as an argument while performe a connection because it's when PDU lenght is negociated.
```
clientInstance = client.Client()
clientInstance.connect("127.0.0.1", 0, 0, 102, pduSize)
```

Sorry I don't know how to do a pull request, so here is the code :

```
...
from ..types import RemotePort, PDURequest, wordlen_to_ctypes, block_types
...

@error_wrap
def connect(self, address: str, rack: int, slot: int, tcpport: int = 102, pduLength: int = -1) -> int:
    """Connects a Client Object to a PLC.

    Args:
        address: IP address of the PLC.
        rack: rack number where the PLC is located.
        slot: slot number where the CPU is located.
        tcpport: port of the PLC.
        pduLength: ...

    Returns:
        Error code from snap7 library.

    Example:
        >>> import snap7
        >>> client = snap7.client.Client()
        >>> client.connect("192.168.0.1", 0, 0)  # port is implicit = 102.
        >>> client.connect("192.168.0.1", 0, 0, 102, 960)  # 960 is the max PDU length for s1500 PLC.
    """
    logger.info(f"connecting to {address}:{tcpport} rack {rack} slot {slot}")

    self.set_param(RemotePort, tcpport)

    if pduLength != -1:
        self.set_pdu_length(pduLength)

    return self._lib.Cli_ConnectTo(self._s7_client, c_char_p(address.encode()), c_int(rack), c_int(slot))

# very redundant :(
def set_pdu_length(self, pduLength) -> int:
    """
    ...

    Returns:
        Error code from snap7 library.
    """
    return self.set_param(PDURequest, pduLength)

def get_pdu_length(self) -> tuple:
    """Returns info about the PDU length (requested and negotiated).

    Returns:
        PDU length.

    Examples:
        >>> client.get_pdu_length()
        (960, 480)
    """
    logger.info("getting PDU length")
    requested_ = c_uint16()
    negotiated_ = c_uint16()
    code = self._lib.Cli_GetPduLength(self._s7_client, byref(requested_), byref(negotiated_))
    check_error(code)
    return (requested_ .value, negotiated_.value)
```
#55
Welcome to make the code cleaner.

lupaulus, 2024-06-24:
Why do you need to extend the PDU size ?
I can be interesting to use your ctor but it need some rework, I will look at it next week.
Moreover, it differ from the base snap7 ctor behavior not sure it will be accepted !

lupaulus, 2024-06-29:
Ok,
Normaly the PDU size is negotiated with the PLC, As for my knowledge a 1500PLC need to negotiate with HIS size (960).
What is your CP paramaters ? Do you change something on TIA Portal about the size of the PDU, if yes, please share some screenshots?

CTor : It's just shorthand for "constructor" - and it's what the constructor is called in Intermediate Language, too.

luewh, 2024-06-30:
> Ok, Normaly the PDU size is negotiated with the PLC, As for my knowledge a 1500PLC need to negotiate with HIS size (960). What is your CP paramaters ? Do you change something on TIA Portal about the [...]

CP parameters ? I just did a regular snap-python db_read(). On plc side just normal security configuration to allow snap7 communication.

ok about Ctor.

lupaulus, 2024-06-30:
CP stands for Communication Processor, it is the "component" that handle the S7 protocol before processing (with the params like type of packet/Protocol, PDU, ...), historicaly it is to have the flexibility to handle other protocols with adding an CP Card (PROFIBUS-DP, OPC, OPCUA, PROFINET), like the CP-1542 to handle PROFIBUS DP on the 1500 Family. 

If you don't use any CP card, you use the built in CP "component" from your 1500 PLC, this "component" negotiated with your external device. They commicate through S7 Protocol in your case,  and will define the PDU size of their communication.

It is know issue in the low-lewel base library, [see the answer of the maintainer of snap7](https://sourceforge.net/p/snap7/discussion/bugfix/thread/7ec0aaed/)

Your code is working but it is more like a "hot fix" until there is a change in the base snap7 package

luewh, 2024-06-30:
> CP stands for Communication Processor, it is the "component" that handle the S7 protocol before processing (with the params like type of packet/Protocol, PDU, ...), historicaly it is to have the flexibility to handle other protocols with adding an CP Card (PROFIBUS-DP, OPC, OPCUA, PROFINET), like the CP-1542 to handle PROFIBUS DP on the 1500 Family.
> 
> If you don't use any CP card, you use the built in CP "component" from your 1500 PLC, this "component" negotiated with your external device. They commicate through S7 Protocol in your case, and will define the PDU size of their communication.
> 
> It is known issue in the low-lewel base library, [see the answer of the maintainer of snap7](https://sourceforge.net/p/snap7/discussion/bugfix/thread/7ec0aaed/)
> 
> Your code is working but it is more like a "hot fix" until there is a change in the base snap7 package

??? I dont see any issue about negociating PDU size...

lupaulus, 2024-07-01:
The negotiated PDU Size is send by the PLC, if they ask for 480, the PLC will accept it because it has the capability to do it.

If you want more you need to declare it with a `SetParam` request, but, for me the 480 bytes values was for the old CPU 300/400, today the 1500 can do more, but the library didn't change this default value when negotiating with 1500 CPU's

luewh, 2024-07-01:
> The negotiated PDU Size is send by the PLC, if they ask for 480, the PLC will accept it because it has the capability to do it.
>
> If you want more you need to declare it with a `SetParam` request, [...]

Ok I understand but no where says that the default PDU size sould be the maximum.

lupaulus, 2024-07-01:
For sure, but at the time the library was written it was an high value to negotiate with the CPU to receive a less or equal value. For my opinion, with the 1500, it need to increase the size of PDU, because of larger objects too. It is an endless question, but the library will not change his behavior.

gijzelaerr, 2024-07-04:
hi! thanks for opening a ticket. 

i don't see a huge advantage in the change you propose, it would eliminate the need for setting the parameter manually, but then there are 15 parameters we could add to the initialisation function, adding them all doesn't improve the situation.

Or am I missing something?

luewh, 2024-07-04:
> hi! thanks for opening a ticket.
>
> i don't see a huge advantage in the change you propose, it would eliminate the need for setting the parameter manually, but then there are 15 parameters we could [...]

Yeah you are right, I just thought that out a week after.
But the ability to change the PDU size isn't documented anywhere and I think it's a commun need for S7-1500 user, open an issue juste for that is worthy.
Let me close this issue.

closed 2024-07-04 by luewh