gijzelaerr/python-snap7 #550: String storage format for the Siemens S7-200 SMART series

lituo666, 2024-11-09:
Can it support the string reading and writing of the S7-200 SMART series PLC, because the string storage format of this series of PLC is different, [the official document link](https://assets.new.siemens.com/siemens/assets/api/uuid:aa045b50-b9f4-4e46-a4c4-ca882c5f00ec/s7-200-smart-system-manual-en-us.pdf), screenshot is as follows:

[image]

Thank you so much

Translated into English by Google, please forgive me for the incoherence of the sentence.

lupaulus, 2024-11-09:
[image]

This project use snap7 that have partial compatibility with the S7 200, the function need to be tested

lituo666, 2024-11-12:
I tested it, and the four areas I, Q, M, and V can be read and written, which basically meets my current needs.

gijzelaerr, 2024-11-12:
so it already works?

hoilung, 2024-11-27:
Remember to set set_comnection.type (3), otherwise there can only be one connection