gijzelaerr/python-snap7 #566: New library for using TIA portal *.db exports to map bytes to python variables

tdamsma, 2025-07-14:
I have just released a library specifically to do this: https://github.com/CEAD-group/s7datablock

It takes a path to a *.db file as input as input, and then allows manipulating the variables like a python dict, or it can print the variables in a TIA portal inspired table. For communication patyon-snap7 is used, so it is really an add on.

[image]

tdamsma, 2025-07-17:
So these PLCtags refer to the global memory, not a particular DB. I haven't needed that kind of functionality. Would you suggest to integrate that functionality?

lupaulus, 2025-07-17:
Here is the format for the DB :

```
  <Tag addr="%DB1.DBX 1.7" type="BOOL" remark="">SIMATIC 400.DB1.Test15</Tag>
```

lupaulus, 2025-07-17:
I wrote a software to do an export from a tia project

gijzelaerr, 2025-12-15:
Thanks for letting us know @tdamsma. But since this doesn't seem a problem with python-snap7 i'm going to close this issue. Or did you propse to include this in the lib?

thanks!